me-no-dev/RasPiArduino #2: BCM2835 registers are not correct for Pi 2

nkolban, 2015-12-18:
The register offsets found in bcm2835_register.h are not correct for the Pi2.  The Pi 1 peripherals are based at 0x2000 0000 while for the Pi 2 they are 0x3f00 0000.

me-no-dev, 2015-12-18:
see in Arduino.c at the bottom :)
```
//maps pi2 offset + 0x1F000000 => 0x3F000000
if(map_registers((getBoardRev() == 0xa01041)?0x1F000000:0)) return 1;
```

me-no-dev, 2015-12-18:
maybe you need to change this to what your value of getBoardRev is, then sketch will fire

nkolban, 2015-12-18:
Gotcha will try that.

nkolban, 2015-12-21:
It seems that the values for the revision as retrieved by getBoardRev are bit encoded ... I think we need to examine the algorithm to determine if this is a Pi2 or not.  See the following:

https://github.com/AndrewFromMelbourne/raspberry_pi_revision

me-no-dev, 2015-12-29:
it seems that everything above 0x15 uses the new algorithm. I can filter those and get more info for the board. Maybe have a struct containing that info and fill it for older boards as well.
Good find!

nkolban, 2016-02-03:
It also appears that the BoardRev returned for a Pi Zero (at least my instance) is:

0x00900092

This has been "hacked into" the `wiring.c` file to allow clean execution on the Pi Zero ... at least until we get this item further designed.

nkolban, 2016-02-03:
The Pi Project found here ... https://github.com/AndrewFromMelbourne/raspberry_pi_revision  was included in the PiDuino project to provide an architected solution for better determination of board identity.

closed 2016-05-06 by me-no-dev