me-no-dev/RasPiArduino #5: README.md is a bit short

probonopd, 2015-12-22:
Would be great to read what this project is all about. Can it be used to run Arduino sketches on the Raspberry Pi? If so, how?

nkolban, 2015-12-22:
Howdy ... see the Wiki ... the project is new on Github and we are still finding our feet ... but YES ... that is exactly the end goal ... run Arduino sketches on the Pi.  The code seems pretty stable right now and appears to work with the tests given.  Welcome to the club.

me-no-dev, 2015-12-23:
@probonopd some new info is available but the project is in really early stage.
If you have a toolchain for raspberry pi, you can give the updated README a try and see if you can get ArduinoIDE to compile and upload/run sketches to the PI