me-no-dev/RasPiArduino #10: Constructors in C++ classes called before init() called

nkolban, 2015-12-23:
In our project, there is a method called init() which sets up the global environment including registers for the BCM2835.  If we link in a sketch, if the sketch has global variables which are instances of C++ classes then their constructors run ... which assumes the envrionment is ready .... but it isn't as the constructors run before main() is given control.

For example:
```
LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

void setup() {
  printf("In setup!\n");
  // set up the LCD's number of columns and rows:
 /lcd.begin(16, 2);
  // Print a message to the LCD.
 /lcd.print("hello, world!");
}

void loop() {
  // set the cursor to column 0, line 1
  // (note: line 1 is the second row, since counting begins with 0):
  lcd.setCursor(0, 1);
  // print the number of seconds since reset:
  lcd.print(millis() / 1000);
}
```

Notice the constructor at the top for the instance of the LiquidCrystal object.

me-no-dev, 2015-12-23:
yeah. That is somewhat a "known" limitation when running on computers
i'm not really sure that the object initialization can be postponed

nkolban, 2015-12-23:
Added some words to the Wiki on warnings and potential circumventions ... https://github.com/me-no-dev/RasPiArduino/wiki/Changes-needed-to-sketches

nkolban, 2015-12-24:
I ran some tests using the GCC function declaration attribute called "constructor" and it seems to work perfectly.  It would appear that we can initialize the registers BEFORE main is invoked and BEFORE the constructors for C++ classes are invoked.

eg.

```
__attribute__((constructor)) void myGlobalInit() {
  pGlobalPtr = (char *)"Hello World";
}
```

me-no-dev, 2015-12-24:
nice! will give it a go

nkolban, 2015-12-24:
Lets make sure we keep in synch ... I've been experimenting with this one and have found some interesting results ... lets make sure we chat so we don't miss something that the other may find.

nkolban, 2015-12-24:
Code changes made and tested.

closed 2015-12-24 by nkolban