me-no-dev/RasPiArduino #11: Use gdb to debug a sketch

nkolban, 2015-12-24:
In the Pi environment we have access to the gdb source code debugger which can be run locally or remotely.  It would seem to be possible to run gdbserver on the Pi and gdb on the PC to attach a remote debugger to the Pi to debug the execution of the sketch.  Quite how we will accomplish this is not yet known ... this issue is to track progress on that possibility.

nkolban, 2016-02-06:
A first pass at implementation has been performed.  This implementation adds a new option in the Tools menu called `Enable gdb`.  The option has two menu choices which are:

* `Enabled`
* `Disabled` (Default)

When enabled is selected, the code is compiled with an extra compilation option ... namely the `-g` flag.  This causes the code to be compiled with additional symbolic information included within it that will allow the GDB debugger (and variants) to be used to source code debug the Sketch.

Here is an example of the new tools menu:

[image]

For more information on compiling with debugging enabled, see:
* [GDB: The GNU Project Debugger](https://www.gnu.org/software/gdb/)
* [Compiling for debugging](https://sourceware.org/gdb/current/onlinedocs/gdb/Compilation.html#Compilation)

nkolban, 2016-02-06:
Initial testing seems to show something not working as expected.  When I place a breakpoint in the code in loop and run the sketch, sure enough we break ... but when I "next", it is almost like a second thread has started and we are toggling between TWO threads of the solution.  I need to consult with @me-no-dev to demonstrate the issue and see if he has any notions of what might be happening.

me-no-dev, 2016-02-06:
try with removing the optimization (-O3 currently). Maybe stuff get's lost there.

nkolban, 2016-02-06:
After removal of the -O3 compilation optimization flags as suggested by @me-no-dev, the gdb debugging now seems to work correctly.  More testing is obviously required but so far no issues.

closed 2016-02-07 by nkolban