me-no-dev/RasPiArduino #14: Random functions not working correctly.

nkolban, 2015-12-25:
The Arduino API provides for functions called random().  These are not yet implemented in our project.

see: https://www.arduino.cc/en/Reference/Random

nkolban, 2015-12-25:
Provided an implementation based off Arduino originals.

nkolban, 2015-12-25:
After further study, we ran into a problem.  The Linux function random() is part of the standard Linux libraries.  Arduino, which is a C++ environment, provides overloads.  This is ok, we can simply create further random() functions as long as they have different signatures.  However, in our project, we have a couple of C (not C++) files ... specifically `arduino.c` and `console.c`.  The C language does not permit function overloading and hence we fail.   The immediate choice seems to be to make `arduino.c` and `console.c` as C++ files by changing their file suffixes to `.cpp`.  Testing seems to show this working (so far).

me-no-dev, 2015-12-29:
we got this fixed right? I'm closing it

closed 2015-12-29 by me-no-dev