me-no-dev/RasPiArduino #17: Logic can infinite loop while using Wire API incorrectly.

nkolban, 2015-12-26:
I recently came across a sketch that had coded:

```
Wire.beginTransmission(0x77);
Wire.receiveFrom(0x77, 1);
...
```

In principle, this is bad logic as we are asking for the start of an I2C write and then immediately asking for the star of an I2C read... however, despite being bad, it shouldn't crash (lock up) the logic.   When `Wire.receiveFrom` is called in this example, it locks up and does not return.  Nothing is transmitted on the I2C bus.

me-no-dev, 2015-12-26:
This is because of my thread safety implementation in cases whan you have threads running and two threads want to write to the Wire bus. Need to investigate that a bit further and maybe try to distinguish the threads and only when there are two or more threads trying to access it, then it should lock in between (the second thread that wants to gain access)

me-no-dev, 2015-12-29:
I removed that functionality and I'm closing this ticket

closed 2015-12-29 by me-no-dev