me-no-dev/RasPiArduino #19: Determine the ARDUINO_ARCH_XXX for the project

nkolban, 2015-12-26:
What will the ARDUINO_ARCH_XXX value be for our project?   Possible considerations are:

* `ARDUINO_ARCH_ARM`
* `ARDUINO_ARCH_RASPI`
* Other?

me-no-dev, 2015-12-26:
it's an interesting question, because it should usually define the _ARM, but if there is such definition already for ARM based MCUs, it might break some code.
I however see those defined at build time:
ARDUINO_RASPI
ARDUINO_ARCH_RASPIARDUINO 
RASPI

me-no-dev, 2015-12-26:
{build.arch} seems to be based on the folder name. In eclipse I use the git folder which is named RasPiArduino and yields ARDUINO_ARCH_RASPIARDUINO
My Arduino IDE uses "piduino" as a folder name and that yields
DARDUINO_ARCH_PIDUINO
We should not use this define I think. Instead we can use #ifdef RASPI

nkolban, 2015-12-26:
I don't think RASPI will work ... it will have to be of  the form ARUDINO_ARCH_ ....

Here is the thinking .... I am running across arduino libraries that need to know the architecture on which they are running.  In order to play well with others, we need to fall in line behind Arduino conventions and standards ... and the convention is that we use ARDUINO_ARCH_XXX.   Thus it would appear to be ARDUINO_ARCH_RASPIARDUINO.

Now we have to decide if this is the name we want to keep.  My vote would be ARDUINO_ARCH_RASPI.

me-no-dev, 2015-12-26:
the word depends on the name of the folder that holds the project files
if you rename that folder to something else that definition will change.
Seems OK for AVR and SAM because they are supplied by Arduino itself.
I would not trust the folder to be called the same. And I have really rarely seen ARDUINO_ARCH_AVR/SAM to be used in any code

closed 2016-05-06 by me-no-dev