me-no-dev/RasPiArduino #21: Library: LedControl fails after a period of time

nkolban, 2015-12-28:
I am running the LedControl library found here:

https://github.com/wayoda/LedControl

I am using a MAX7219 with an 8x8 matrix.  When I run the sample, it all runs for a period of time and then ... after about 5-10 minutes ... the 8x8 matrix goes dark.  I haven't yet diagnosed the problem ... I don't know if the signals from the Pi have gone wrong or whether I have a faulty MAX7219 ...   The Sketch running on the Pi does not crash or stop running.  If I stop the Sketch and then immediately restart it, the display immediately starts responding for 5-10 minutes ... until it doesn't any longer.

me-no-dev, 2015-12-29:
I looked through the library code and the core functions used and I see nothing wrong
I have no idea what is going on but if you can attach your logic analyzer and see if there is data coming out and if it's the same

me-no-dev, 2015-12-29:
I checked the datasheet and it looks like it can handle 10MHz SPI and the shiftOut method provides 500KHz, so it should be well in specs even if something somehow messes the timing at some point

nkolban, 2016-01-03:
I am no longer convinced that this is a project problem.  I re-implemented in WiringPi and have same issues.  I looked at low level traces using a logic analyzer and all looks good.  Am awaiting delivery of a different 7 segment module to test differently.

closed 2016-05-06 by me-no-dev