me-no-dev/RasPiArduino #26: Will wifi libraries be implemented to allow use of wifi adapter under this project ?

mmiscool, 2016-01-06:
Will wifi libraries be implemented to allow use of wifi adapter under this project ?

nkolban, 2016-01-06:
If not they can easily be implemented.  Please help us to help you ... can you point to the WiFi interface that you want to see implemented.  Specifically, if you can point to what the existing API looks like, that will be a fantastic help.

me-no-dev, 2016-01-06:
Should be pretty trivial. Not sure if it will be WiFi specific per say, maybe just Server/Client classes to facilitate the network as a whole. Shell interface to execute commands will also be added, maybe that will help if change of AP needs to be done :)

me-no-dev, 2016-01-08:
Network libraries are in (the basics that is).
PiServer, PiClient and PiUDP classes follow the signature defined by Arduino, so any library that conforms to those can use them and have a working network.
@mmiscool is there any particular feature that you are interested in?

closed 2016-05-06 by me-no-dev