me-no-dev/RasPiArduino #44: Any chance you could team up with dwelch67

KeithSloan, 2016-02-08:
I would like to see the ability to load Arduino sketches into the Raspberry Pi with out an operating system. It looks like you have done a lot of the work of interfacing with the Arduino IDE and dwelch67 has created standalone boot loaders see https://github.com/dwelch67/raspberrypi. Any chance you could team up?

me-no-dev, 2016-02-08:
yes :) I have also thought much of the bare metal possibilities. It will not at all be hard to convert the lib to talk bare metal. I already have some nice sources on how to access the USB as well. Will need to look into it more to be able to determine how could the whole thing work together (reload the sketches from IDE).

nkolban, 2016-02-08:
@KeithSloan Many thanks for query.  Would you be able to describe some example use cases where a "Raspbian free" version of PiDuino might be used?  What features in it would you see as beneficial above and beyond the version that is currently available running on Raspbian?

me-no-dev, 2016-02-08:
@nkolban any real-time usage. It is turned into a micro-controller of a sort...
No kernel scheduler to mess up timings and such. It's like Arduino for real

KeithSloan, 2016-02-08:
Hi Neil

I would see the benefit of a bar metal implementation as you don't have 
to worry that your sketch being subject to interrupts.
That is one could be much more assured with things which have critical 
timing.

Should be easier to setup as no need to disable features in Raspbian.

Just seems to me to be a cleaner solution

Keith

me-no-dev, 2016-02-08:
Cleaner for certain situations. Currently you have access to all that linux can offer. Bare Metal is Bare :) there is no linux there. You do your own networking and all. No Shell to execute scripts or other goodies that come with OS. SO they are both for really different situations.
I do agree that there is a way to make the framework work on both Linux and BareMetal, but do see majority of the users benefitting from having the OS in the background.

KeithSloan, 2016-02-08:
Well in your shoes I would look at offering both.

For me if I have sketches developed for Arduino or chipKit pi I would 
definitely choose the bare metal option.

I was under the assumption that one would be restricted to what the 
Arduino API supports so no SPI-Slave etc.
So how does one access the Linux function

me-no-dev, 2016-02-08:
Using this project in it's current state is just like writing a C++ app for any linux distro + access to the BCM peripheral hardware. If you export your libraries from the Pi, you can link to any lib and use it in your code (libusb for example or the libs for interfacing with the camera module). You can also load kernel modules and use them as well.
And I am interested in trying to offer both. Will see what I can do.