me-no-dev/RasPiArduino #46: No command lines are shown when i click the verify button in Arduino IDE

Algorithmss, 2016-06-19:


me-no-dev, 2016-06-19:
You need to enable verbose compilation in ArduinoIDE settings.
[image]

Algorithmss, 2016-06-19:
And also can you help me with this error? 

exec: "C:\\Program Files (x86)\\Arduino\\hardware\\RaspberryPi\\piduino/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++": file does not exist

Algorithmss, 2016-06-19:
I got it to work once then i copied the bin file into my raspberry pi but it says that i cannot run it, it shows me " command not found " error

me-no-dev, 2016-06-19:
do you have the toolchain installed on your computer? If you do and you build, you can export the binary to the sketch folder, then copy to the py, make it executable (chmod +x your_file.bin), then run it like above (it has to be executable for that command to work)

Algorithmss, 2016-06-19:
Okay thanks, let me try that method out :)

Algorithmss, 2016-06-19:
Thanks it works... :D but it just keep on looping and looping and it goes on and on and i can't make it stop

me-no-dev, 2016-06-19:
that is how an arduino sketch works :) Ctrl+C to stop it

me-no-dev, 2016-06-19:
if you follow the readme and do the modifications on the PI, you will be able to upload directly from the PC and the service will take care of killing the previous sketch

Algorithmss, 2016-06-19:
Yup i want to do that too, i will try it out. And also, i can just write the code down in Arduino language right? Or is there a difference?

me-no-dev, 2016-06-19:
no difference :) it is arduino framework. so Wire, SPI and all is there for you to use. Most sketches require usually only pin change for them to work on the Pi

Algorithmss, 2016-06-19:
Oh okayy sounds great, Thanks for the help Mr. I am new to all this and i want to try out projects :)

Algorithmss, 2016-06-19:
Oh and is there any chance i can use this with Blynk app? If yess tell me how :) 
Thanks x)

me-no-dev, 2016-06-19:
what is Blynk app?

Algorithmss, 2016-06-19:
It is basically like a remote app that control your devices, from the app. http://www.blynk.cc/   here's the link to the app website. I am bad at explaining so wanna check it out? it is really cool

Algorithmss, 2016-06-20:
Ohh also, when i installed this, does it still function as a normal Raspberry Pi? Like can i put others code in? or like add some git code inside?

me-no-dev, 2016-06-21:
yes, it's just another program that runs on your PI :) you can still use it as a linux box or whatever you want. Just keep in mind to not use the GPIO pins for anything else while the sketch is running, because it controls the whole pinout