me-no-dev/RasPiArduino #47: Unable to compile in Windows Arduino IDE - ctags_target_for_gcc_minus_e_cpp

nkolban, 2016-07-10:
On a Windows 10 environment, I am attempting to compile a trivial Arduino Sketch.  When I try this, the compilation fails and the error logs shows:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Kolban\Documents\Arduino\libraries" -fqbn=RaspberryPi:piduino:bplus:SerialDestination=disabled,GDBDebug=disabled -ide-version=10609 -build-path "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Kolban\Documents\Arduino\libraries" -fqbn=RaspberryPi:piduino:bplus:SerialDestination=disabled,GDBDebug=disabled -ide-version=10609 -build-path "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino"
"C:/SysGCC/Raspberry/bin/arm-linux-gnueabihf-g++" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10609 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\variants\bplus" "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\sketch\Blink.ino.cpp" -o "nul"
"C:/SysGCC/Raspberry/bin/arm-linux-gnueabihf-g++" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10609 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\variants\bplus" "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\sketch\Blink.ino.cpp" -o "nul"
"C:/SysGCC/Raspberry/bin/arm-linux-gnueabihf-g++" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10609 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino" "-IC:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\variants\bplus" "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\sketch\Blink.ino.cpp" -o "C:\Users\Kolban\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
exit status 1
Error compiling for board RaspberryPI B+/2.
```

It has been 6 months since I played with this environment and have re-installed all tools to latest levels.  I was attempting to track down an issue reported by user "zaukkoo" in the forum thread found here ... https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=134928  where he too reported the exact same symptom ... see his post 2016-07-07.

me-no-dev, 2016-07-10:
interesting... why is it compiling to output "nul"? What IDE version are you using? As far as I remember, I tested the 64 bit win on win10 and was fine (6 months ago), have since used it on my Mac, but have not tested with latest IDE

nkolban, 2016-07-10:
Am using the latest 1.6.9 IDE freshly downloaded.

blissillour, 2017-02-20:
I get the same error. Any suggestion on how to process to fix it ?