me-no-dev/RasPiArduino #49: How we can run arduino generated .bin file on raspberry pi 3?

jawadg, 2016-08-03:
we use raspi arduino framework https://github.com/me-no-dev/RasPiArduino when we upload or run .bin file on raspberry pi 3 error occur UNKNOWN-REVISION 0x00000000. the link https://github.com/me-no-dev/RasPiArduino library successfully run on raspberry pi 2.

me-no-dev, 2016-08-03:
sorry, I do not own pi3 yet and have not tested the framework with it.

prothan, 2016-08-05:
Get error UNKNOWN_REVISION: 0x00A02082 when I try to run binary on pi3. I also have pi 2, worked good on that. Really would like to try to get this working on pi3. Any thoughts on what the issue might be?

PocoCraft, 2016-11-21:
I also recieve the 'UNKNOWN_REVISION: 0x00A22082' error on my Pi3..
Although I'm not sure if this is an active project anymore so any update on this issue would be appreciated.

MovLab2, 2016-11-22:
I too get this error. It is for using pi3. /proc/cpuinfo=
Hardware : BCM2709
                                                                                     Revision   : a22082
                                                          works with          Revision   : a21041       = Pi2  
This is such a good way.    Question for me-no-dev,    Do You intend to get this working for us?
Is there a workaround? Hope there is something.. Thanks if you can...

me-no-dev, 2016-11-22:
yes, I surely intend to get it to run on all PIs out there, but I am really currently preoccupied with work on the ESP32-Arduino core :( have not even had time to open the browser and purchase PI3. As far as I know there are no changes in registers, but I am unaware yet if they follow the PI1 addressing or the PI2. If someone can dig some info maybe I can help get this going on one of your PIs? :)

me-no-dev, 2016-11-22:
Here is some info on what you can do:
1. Error comes from [here](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/wiring.c#L175) because that revision is not known.
2. Revisions supported are defined [here](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/bcm2835_registers.h#L662)
3. Library used to detect the model is [here](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/raspberry_pi_revision.c). Register location is also there. Source repository for it is [here](https://github.com/AndrewFromMelbourne/raspberry_pi_revision)

So maybe you can replace with updated detection lib and add the mapping for PI3. The rest should work from then on.

me-no-dev, 2016-11-22:
ok, nevermind :) I pushed pull the latest commit and give it a go. It might work :)

MovLab2, 2016-11-22:
It's Running Perfect now.. Thanks so much.  My test compiles and runs in the pi3 now .. 
Great Job..     Now I can continue with my project :)))

PocoCraft, 2016-11-22:
I can confirm the test sketch I tried running is now working after updating the RasPiArduino utilities on the Raspberry Pi 3.

Thanks a ton! :D

dvelopin, 2017-01-08:
What's the best way to use sensors? Like if on Arduino you said to use pin 4, would pin 4 still be the Pi's data pin 4 (physical pin 7)?

mega-hz, 2018-02-03:
Hello, great projekt! How can i get it running on a bananaPi ? I get an error on it: 
root@HOME-SYS:/home# sudo ./test_pi.bin 
UNKNOWN_REVISION: 0x00000000
Would be great if you could help me! Wolfram.﻿