me-no-dev/RasPiArduino #51: Using port other than ttyAMA0 for console

jetpax, 2016-09-06:
Hi Neil

I have a SC16is750 I2C UART installed on my Pi (as /dev/ttySC0) that I'd like to use as a serial port in the RasPiArduino environment.

Can you give me some pointers as to how I could go about doing that?

I see the TTY.cpp library, which seems to refer to /dev/tty, but don't understand how that relates to HardwareSerial.cpp which refers to actual RPi GPIO?

nkolban, 2016-09-06:
I'm still studying this code myself.   @me-no-dev is the master of the code.  I looked at the TTY class and was surprised to see that the constructor of it expects a "numeric" ... so creating an instance of it with "17" will open /dev/tty17 ... great ... but if we have a TTY called /dev/ttySC0, we can't seem to use TTY!!   That seems a trifle odd.

me-no-dev, 2016-09-07:
The tty lib is actually to control the console tty that you see when you plug in a monitor/tv. With it you can write to it like you are using a keyboard

jetpax, 2016-09-07:
I see, and thanks, both for creating a cool project, and for replying.

So as tty lib exposes almost the same interface as serial, i suppose i
could just create another serial port using tty as a base, eg
with termios???

Would be good if we could generalise it to use say ttyUSB0  etc as well, so
that Arduino apps could use whatever serial ports Linux knows about...

Ill have a play

BTW I see other forks have taken your project forward, are you planning on
merging those?

jetpax, 2016-09-07:
Thanks Neil, thats sort of what im puzzling over, it looks like it should
be possible to use termios instead of accessing the hardware directly , but
i dont know enough about it yet to be sure...

And thanks for your books, esp the ESP8266

me-no-dev, 2016-09-07:
As i am currently employed by Espressif I have little time to work on the pi :) i don't even have pi 3 yet. I accept PRs but none have been sent.

me-no-dev, 2016-09-07:
I would advise you not use the tty lib but rather unload the kernel driver and find arduino lib for your i2c uart and use that to communicate. It will be all fast and good :)