me-no-dev/RasPiArduino #53: Raspberry Pi2 latest version

scargill, 2016-11-22:
Problems,

BLINK sketch - running often kills the Pi's network connection...   just a blink sketch - pin 18 - flashing on and off - first time I've ever had a Pi die on me.

Your software downloaded and setup today.

me-no-dev, 2016-11-23:
could you please tell more about your environment? PI Version, OS used? Anything else running on the PI that might want to talk to the pins? Did the sketch die immediately, or did it blink a few times?

scargill, 2016-11-23:
Ok, latest Pi Debian (Pixel) - up to date as of a day or two ago - Node-Red running.   I've now rebooted it with nothing running but your damien to make it available - and it the Ethernet connection is dying after a short time of sitting doing nothing.  Now - when running the BLINK sketch - the sketch continues to operate no problem - it is the Ethernet connection which stops. The Pi2 has both WIRED and WIRELESS connections (192.168.0.127 an 192.168.0.126) and both stop working.

me-no-dev, 2016-11-23:
this is really interesting... what does my code have to do with USB/Network (your USBs probably stop too right?). Also I do have a PI2 on my desk that runs a sketch, LAN and WiFi for over 6 months now, so whatever is happening must be new... still RasPIArduino has nothing to do with USB and LAN :(

thebreadboard, 2016-11-30:
This has also shown up on PI3, not just PI2

me-no-dev, 2016-12-08:
I loaded pixel on pi zero and setup everything for Arduino. Difference I saw was the way I had to disable serial (raspi-config), but other than that everything else worked fine. Also loaded over dwc2 RNDIS network and serial, no problem. Still, this framework has nothing to do with Ethernet or Wireless... it uses standard C api to talk to outside, just like any other application.