me-no-dev/RasPiArduino #54: Serial on PI2B using PIXEL version of Raspian

thebreadboard, 2016-11-23:
Tried the RasPiArduino on a Pixel version of Raspbian and it fails every time I try to use the ttl serial (Pins 8,10). I2C and Digital IO still seems to be ok.

I reverted back to Raspbian from about may 2016 and everything worked great so it looks like the RPI folks have changed something.

thebreadboard, 2016-11-23:
I tried to reproduce a working serial on the May "Raspbian" and also did further tests on a PIXEL version.
If I use Node-red and access the serial on pins 8/10 everything works ok. (Using Node-Red serial module)
The second I run a sketch that tries to use the serial.println.... it stops working, even if i kill the arduino sketch and try the node-red again, it still does not work. The sketch does not write to the serial ttl pins. If i look in the run/sketch log it shows the text that was supposed to go out the serial.

If I run the resulting bin file directly from a command prompt on the PI it outputs the serial to the console, not the ttyAMA... port.

 I have to reboot to get the node-red tty access working again, simply stopping the sketch is not enough

 I also tried this on the earlier version of Raspbian again, I must have done something off the first time around and made it work, but following the instructions on a new PI2 and New imaged SD card, it does the same thing as with the PIXEL version of raspbian. I do not recall what I did the first time but the raspbian image had other installs tried to get an arduino sketch to run prior to using this version so it is possible one of them had changed something. Using a brand new SD image this process fails with the serial, and as Pete Scargill has reported, there are other stability issues needing to be addressed.

I hope you can fix this as I was hoping to use this version of RasPiArduino in my training videos on my youtube channel.

me-no-dev, 2016-11-23:
All you explain is really really strange. This lib talks to the Serial directly, bypassing any kernel or other code. Thera must be a race somewhere with the drivers or something, but even then there should be garbage on the serial rather than the failures you explain.
I have been so preoccupied with work that I didn't even know there is a PIXEL version of raspbian?

I'll grab  pi3 these days and run through the setups again. Which PI do you use by the way?

thebreadboard, 2016-11-23:
I am using a Raspberry PI 2, model B V1.1

 

I get same issue with Pixel version and with May 2016 version of the OS. Also once the app finishes it does not restore the serial chip back to operation from the serial0 or ttyAMA0 unless I reboot the OS.

 

Thanks for the help, this looks like a great implementation, I would just love for the serial to work.

 

Peter

thebreadboard, 2016-11-30:
PI 2 and PI 3, same issue

me-no-dev, 2016-12-08:
So I got some PIXEL goodness on pi Zero and in raspi-config -> advanced there was an option to disable the serial console, which made the rest work fine. Can you try that?

someMEXICANz, 2020-04-08:
I could use some help, I'm working with a RPI4 with the latest raspbian and I'm been trying to get serial to run properly when using a serial to USB adapter. Everything seems to work fine up until I try to monitor it via Arduino serial monitor or putty, it will output backwards question marks and random characters. I've made sure baud rates are set properly and tried different speeds. If I send data from the monitor nothing shows and serial.read() spits out a -1, so no data. Maybe I'm missing a step, but the issue seemed similar to this thread any help would be much appreciated.

thebreadboard, 2020-04-12:
What adapter are you trying to use. I have used some in the past with no issues.

 

Regards

Peter

someMEXICANz, 2020-04-12:
This is the one

http://www.dsdtech-global.com/2017/07/dsd-tech-usb-to-ttl-serial-converter.html?m=1

Thanks,

thebreadboard, 2020-04-12:
So here are some questions

 

The USB to serial adapter is plugged in the PC or the PI ??

The Serial end is going to the PI RX/TX pins on the GPIO header of the PI or to the TX/RX of the Arduino ??

Your connecting the Arduino to the PI ?

 

I guess what im asking is how is this all connected, a quick diagram with pins used would help

 

Some things to note

 

When you connect one serial to the other, you need to cross the TX and RX, so TX of one goes to RX of the other

 

The Arduino is a 5V device, the PI is a 3V3, what is the voltages coming out of the USB adapter TX pin ?

 

Regards

Peter

 

http://www.dsdtech-global.com/2017/07/dsd-tech-usb-to-ttl-serial-converter.html?m=1