me-no-dev/RasPiArduino #55: which pre-emptive multithreading libs ?

shiftleftplusone, 2016-12-04:
hello,
which pre-emptive multithreading libs are working with your RasPiArduino framework?
Are there code examples available?

me-no-dev, 2016-12-04:
Compiled binaries are standard Linux executables, so you can use basically anything that is available on linux (given that you have the required libs and headers on the build machine)

shiftleftplusone, 2016-12-04:
ok, but IIUYC, I am crosscompiling from a PC based Arduino IDE to the  Pi code, 
so does the Ardiono IDE know what do if I 
#include <pthread.h>
at the top and then start the threads

```

void* thread0 (void* ) {       //  
    while(1) {    		
    //
    }
    return NULL;
}
void* thread1 (void* ) {       //  
    while(1) {    		
    //
    }
    return NULL;
}
void* thread2 (void* ) {       //  
    while(1) {    		
    //
    }
    return NULL;
}

void setup(){
    pthread_t thread0, thread1, thread;
    struct  sched_param  param;

    pthread_create(&tid0, NULL, thread0, NULL);    //  
    param.sched_priority = 20;
    pthread_setschedparam(thread0, SCHED_RR, &param);
   
    pthread_create(&tid1, NULL, thread1, NULL);    //  
    param.sched_priority = 30;
    pthread_setschedparam(thread1, SCHED_FIFO, &param);

    pthread_create(&tid2, NULL, thread2, NULL);    //  
    param.sched_priority = 30;
    pthread_setschedparam(thread1, SCHED_FIFO, &param);

   // *snip*
}

void loop() 7
//
}
```
AFAIK, Arduino #includes only files located in /program files (x86) \Arduino\libraries in a proprietary folder of the same name as the lib header.
For the Scheduler libs (Arduino Due) that works, also for Babix and RTOS48 on ARMs and AVRs -
but are you sure that POSIX pthread will work that way,compiled on a PC,and then uploaded?

me-no-dev, 2016-12-04:
yes :) As a matter of fact, RaspiArduino has threads already running underneath ;) Serial events, monitor socket and some other things have their own threads. And there already is some api added for pthreads in [Arduino.h](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/Arduino.h#L160)

shiftleftplusone, 2016-12-04:
that's awesome.
Currently I am still a little scared about the monster installation guide  (a install.exe which does all what's  needed would be very much appreciated though ;)  )  - but perhaps I can persuade my self and give it a try then, hopefully soon. ;)

me-no-dev, 2016-12-04:
There are some things that can not (should not) be done in an installer, because they are external dependencies that have their own licenses and all (it's actually one thing and that is the toolchain for windows). On the PI itself, many things are optional and depend on each user's preferences. I will at some point make some sort of script to automate the things on the PI at least for the bare minimum.

shiftleftplusone, 2016-12-04:
thanks,   that's highly  appreciated!  8)