me-no-dev/RasPiArduino #56: Use EthernetServer with Pi network

moelski, 2016-12-14:
Hi !

I tried to compile the webserver example which works fine. But after looking into the libs I saw that it is all based on the W5100 chip - so the standard ethernet breakout for the arduino. 
The Pi has embedded network support so can you give some advice how to use the EthernetServer class with the Pi network interface?

Dominik

me-no-dev, 2016-12-14:
Sure :)
The PI uses the Bridge classes to provide networking, file access and more.
You can find it [here](https://github.com/me-no-dev/RasPiArduino/tree/master/libraries/Bridge/src)

moelski, 2016-12-14:
Hi !

Shame on me ... 
Will give it a try. Thx for your help.

moelski, 2016-12-14:
Hi again, 
you can Close this issue. The Bridge Classes work great.

me-no-dev, 2016-12-14:
good :)

closed 2016-12-14 by me-no-dev