me-no-dev/RasPiArduino #57: How to cast to String?

moelski, 2016-12-14:
Hi !

Just try to get the ESP8266WebServer running with RasPiArduino. It works but there are some small Problems. One is the casting of int to string for example. 
This is the some Code from the webserver: 
`void PiWebServer::_prepareHeader(String& response, int code, const char* content_type, size_t contentLength) {
    response = "HTTP/1."+String(_currentVersion)+" ";
    response += String(code);
...}`
The casting of _currentVersion (which is uint8_t) results in an unreadable character. 

I found a solution which works: 
`  char buffer[256];
  sprintf(buffer, "%d", x);`

But maybe there is a more elegant and easier solution?

Dominik

me-no-dev, 2016-12-14:
this sounds rather like bug in String

moelski, 2016-12-15:
Could you have a short look if this issue can be fixed ?

moelski, 2016-12-16:
Hi !
I have enhanced the WString class and now all string related stuff is working. 
I have also make a "PiWebServer" class based on the ESP8266 stuff. 
Would love to make a pull request - but I can´t ... 

So How can I contribute to this project ?

me-no-dev, 2016-12-16:
you fork the repository, add your changes there and make a PR to this repo :)

moelski, 2016-12-16:
Hi !

I think this topic can be closed. The PR #58 from me fixes all my issues.

closed 2016-12-16 by moelski