me-no-dev/RasPiArduino #62: Compiling Error When Using ArduinoFFT Library

amd940, 2017-01-23:
Hello, I'm having issues getting my sketch to compile into a .bin file when using the [ArduinoFFT Library](http://wiki.openmusiclabs.com/wiki/ArduinoFFT) from Open Music Labs. This is the first library I've been having troubles with, every other one compiles just fine, so I'm not sure if this is a problem with the library or the cross-compiler. These are the error messages I get:

```
Arduino: 1.8.1 (Windows 10), Board: "RaspberryPI B+/2, Disabled, Disabled"

In file included from C:\Users\Justin\Documents\Arduino\sketch_jan23a\sketch_jan23a.ino:3:0:

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h: In function 'void loop()':

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:1336:4: error: unknown register name 'r23' in 'asm'

   );

    ^

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:1336:4: error: unknown register name 'r22' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:1336:4: error: unknown register name 'r31' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:1336:4: error: unknown register name 'r30' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:1336:4: error: unknown register name 'r20' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:805:4: error: unknown register name 'r31' in 'asm'

   );

    ^

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:805:4: error: unknown register name 'r30' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:805:4: error: unknown register name 'r27' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:805:4: error: unknown register name 'r26' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:805:4: error: unknown register name 'r20' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r31' in 'asm'

   );

    ^

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r30' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r27' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r26' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r25' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r24' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r23' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r22' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r21' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r20' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r19' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:713:4: error: unknown register name 'r18' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r20' in 'asm'

   );

    ^

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r19' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r18' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r31' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r30' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r27' in 'asm'

C:\Users\Justin\Documents\Arduino\libraries\ArduinoFFT3/FFT.h:947:4: error: unknown register name 'r26' in 'asm'

exit status 1
Error compiling for board RaspberryPI B+/2.
```

I've tried figuring out why this particular library will not compile for the Raspberry Pi but I've not had any luck. It will compile without errors for an Arduino board. I didn't include the code because it doesn't seem to make any difference what's in the code, I just keep getting the same errors. Anyways, thanks so much for any help you can provide.

me-no-dev, 2017-01-23:
the library you linked is strictly AVR library written in assembler. It could never work on anything other than AVR Arduino

amd940, 2017-01-23:
Okay, I did not realize that. I'm a newbie with Arduino. Thanks for the info. I'll mark this issues as closed.

closed 2017-01-23 by amd940