me-no-dev/RasPiArduino #63: Getting bin file to run on Pi zero w

cjscottcjscott, 2017-04-25:
I read the thread on getting to run on Pi3. Any chance of updating for executing bins on Pi Zero w?

me-no-dev, 2017-04-26:
Uhmmm is it not working? I'm hoping to get one soon, but I expected it to already work. What is the error you are getting?

microbrewer, 2017-06-11:
Hello, I have the same problem. Can't seem to run a sketch on the Zero W. The issue seems ti be similar to the one in #49. I get the following error:
`UNKNOWN_REVISION: 0x009000C1`

This is the content of /proc/cpuinfo on my Zero W v1.1:
`# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000dd0865c4`

So 9000c1 seems to be an unknown revision. Hope that helps. I would very much like to get this to work. :)

microbrewer, 2017-06-16:
Ok, so I tried to understand what @me-no-dev did in https://github.com/me-no-dev/RasPiArduino/commit/89dd03f76b6b021e950c48733cd159ec4622faeb to support the Pi3, and if I'm not mistaken, all we need to do is update the revision detection lib from https://github.com/AndrewFromMelbourne/raspberry_pi_revision which already supports the Zero W, and modify one line in wiring.c:181

`if (info.model == RPI_MODEL_B_PI_3 || info.model == RPI_MODEL_B_PI_2 || info.model == RPI_MODEL_ZERO || info.model == RPI_MODEL_ZERO_W) {`

since the pin masks on the Zero and Zero W are the same. Am I missing anything?

me-no-dev, 2017-06-21:
Nope! that is exactly what needs to be done

beilpig, 2017-07-10:
Thanks heaps that worked a treat!

RamyRamo, 2023-09-01:
Please, where I found revision detection lib? in  https://github.com/AndrewFromMelbourne/raspberry_pi_revision 
there is no file name wiring.c 
Any help please? I need to use pi zero w board