me-no-dev/RasPiArduino #70: I2C communication with RasPiArduino and SPI

craig42, 2017-09-13:
Hi, 

I have a Raspberry 3 and I use RasPiArduino for I2C communication with this gas sensor : 
http://wiki.seeed.cc/Grove-Multichannel_Gas_Sensor/

And I use this LoRa hardware which communicates in SPI which use this library : https://github.com/Snootlab/lora_chisterapi/blob/master/lib/radiohead/RH_RF95.cpp

When I start RasPiArduino, LoRa hardware can't send and receive data so I think RaspiArduino use SPI...

Is it possible to disable SPI connection in RasPiArduino ?

me-no-dev, 2017-09-13:
it uses interrupts and maybe that is the issue ;) RPI does not really have interrupt vectors

craig42, 2017-09-13:
So, what can I do ? It's not possible to run the two program at the same time ?

me-no-dev, 2017-09-13:
ahh wait... I missread your initial post ;) you access the SPI through another APP and I2C through Arduino. Would it work if you first start Arduino and then RF95?

craig42, 2017-09-18:
No, I tested it. 
But now, I think RasPiArduino is not the problem. I test LoRa Hardware HAT with a simple LED. The LED is controlled by python program, and when I launch LoRa program, I have an error message "Init failed"