me-no-dev/RasPiArduino #72: Unable to connect to 192.168.0.67

dzyk, 2017-09-28:
./arduino
Picked up JAVA_TOOL_OPTIONS: 
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
processing.app.debug.RunnerException: Unable to connect to 192.168.0.67
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:143)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2168)
	at java.lang.Thread.run(Thread.java:748)
Unable to connect to 192.168.0.67

me-no-dev, 2017-09-29:
probably missed a step in install? SSH not installed on the PI?

dzyk, 2017-09-29:
ssh good work... i connect with RPI3 with ssh and vnc

dzyk, 2017-09-29:
install instruction complete all for this project

dzyk, 2017-09-29:
help me, please

me-no-dev, 2017-09-29:
The error comers from Arduino IDE. You can see that it try to connect to that IP through SSH but it fails. Either SSH is down or IP is wrong or ArduinoIDE craps. :)

dzyk, 2017-09-29:
ssh work
IP ping OK
IDE? Mey be... what can i see with IDE? its a panic for me

me-no-dev, 2017-09-30:
ide is compiled java, so there is little you can see. what version is arduino ide?

dzyk, 2017-09-30:
version 1.8.4

dzyk, 2017-10-03:
i connect... ssh is down. you right

closed 2017-10-03 by dzyk