me-no-dev/RasPiArduino #73: my password wrong

dzyk, 2017-10-03:
i type password for scketch upload... in its wrong? what?

dzyk, 2017-10-03:
[https://imgur.com/E3IbKjz](url)
see screhshot

me-no-dev, 2017-10-04:
uses root account so you must set the root password first if you have not done so

dzyk, 2017-10-04:
root and user password wrong for upload((( i used it

me-no-dev, 2017-10-04:
can you ssh root@raspberrypi-ip ? sudo and root passwords are two different things

dzyk, 2017-10-04:
password for root and user identical

me-no-dev, 2017-10-04:
So were you able to ssh as root to your pi? Or did you ssh as pi and then did sudo/su? Did you set the root password yourself? Did you make sure that in sshd.conf root logins are permitted? What did you really try?

dzyk, 2017-10-04:
Password for pi@piduino:12345pass

    Gain root permissions

sudo su

    Enable password login for root

passwd

    enter the new root password twice

--> change to 12345pass

dzyk, 2017-10-04:
pi@piduino:/etc/ssh $ cat ssh_config 

Host *
    SendEnv LANG LC_*
    HashKnownHosts yes
    GSSAPIAuthentication yes

dzyk, 2017-10-04:
pi@piduino:/etc/ssh $ cat sshd_config  
UsePAM yes
X11Forwarding yes
PrintMotd no
AcceptEnv LANG LC_*
Subsystem	sftp	/usr/lib/openssh/sftp-server

dzyk, 2017-10-08:
What can i do for change password?

dzyk, 2017-10-09:
its a default for me(