me-no-dev/RasPiArduino #92: Sketch compiles once, then gives error

kd8bxp, 2018-11-07:
My sketch compiles once, then I get the following error:
```
Arduino: 1.8.5 (Linux), Board: "RaspberryPI B+/2, Enabled, Disabled"

Warning: Board mighty-1284p:mighty-1284p:mighty_opt8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_OPT8
Warning: Board mighty-1284p:mighty-1284p:bobuino doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_BOBUINO
Warning: Board mighty-1284p:mighty-1284p:mighty8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY8
Warning: Board mighty-1284p:mighty-1284p:mighty_int8_opt doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_INT8_OPT
Warning: Board mighty-1284p:mighty-1284p:mighty_opt doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_OPT
Warning: Board mighty-1284p:mighty-1284p:avr_developers doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_AVR_DEVELOPERS
Warning: Board mighty-1284p:mighty-1284p:mighty doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY
Warning: Board mighty-1284p:mighty-1284p:mighty_int8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_INT8
panic: runtime error: index out of range

goroutine 1 [running]:
panic(0x556bc0, 0xc420010110)
	/home/jenkins/go/src/runtime/panic.go:500 +0x1a1
arduino.cc/builder.findIncludesUntilDone(0xc420062400, 0xc420587ad0, 0x556f40, 0xc4204ea0f0, 0xc420428e37, 0x9, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_find_includes.go:321 +0x138f
arduino.cc/builder.(*ContainerFindIncludes).Run(0x64d248, 0xc420062400, 0x41bc3c3c, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_find_includes.go:149 +0x5e4
arduino.cc/builder.runCommands(0xc420062400, 0xc42059bad8, 0x22, 0x22, 0x1, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 +0xcd
arduino.cc/builder.(*Builder).Run(0xc42059bd28, 0xc420062400, 0xc420045d88, 0x562240)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 +0xb9c
arduino.cc/builder.RunBuilder(0xc420062400, 0xc420045d88, 0x64d248)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 +0x99c
/home/lfmiller/.arduino-1.8.5/arduino-builder returned 2
Error compiling for board RaspberryPI B+/2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
If I close the sketch and reopen it, it compiles again.  I wrote a custom library for a robot I am working on, and I think the problem is when I include the library, if I just use my "test" code.
My code can be found here: https://gist.github.com/kd8bxp/dcc6d8c276a49ef31b77032d4862d510
I did change the sketch using the recommended library settings from the wiki (test2.ino) But no change.
Frankly I'm a little lost, and I'm not sure why or what to do to fix this. Any help, any ideas...

P.S. this error only happens with the Raspberry PI board core selected, I can compile over and over again using a Arduino UNO as the board.

Barabba11, 2018-11-07:
One suggestion, be sure the compiler is using the libraries of Raspberry, it happen to me dozen of times it used the wrong ones. Libraries can be in different folders, on hardware folder in Arduino program folder, or under your documents Libraries. Sometimes you may have duplicate .h files in other folders. Check really carefully.

kd8bxp, 2018-11-08:
@Barabba11 I've run into that issue before with different cores/libraries being named the same as something already installed, ect. But If this was true for this issue, why would the test sketches compile once (?) 

I can compile other sketches for the Raspberry PI over and over again as well.

I found a similar issue that I have with the ChipKit board core, but I couldn't find an answer to my problem.

Compiled with verbose turned on so I could see which libraries are being used:
```
Compiled (once) Compilation verbose output on:

/home/lfmiller/.arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/lfmiller/.arduino-1.8.5/hardware -hardware /home/lfmiller/.arduino15/packages -hardware /home/lfmiller/Arduino/hardware -tools /home/lfmiller/.arduino-1.8.5/tools-builder -tools /home/lfmiller/.arduino-1.8.5/hardware/tools/avr -tools /home/lfmiller/.arduino15/packages -built-in-libraries /home/lfmiller/.arduino-1.8.5/libraries -libraries /home/lfmiller/Arduino/libraries -fqbn=RaspberryPi:piduino:bplus:SerialDestination=disabled,GDBDebug=disabled -ide-version=10805 -build-path /tmp/arduino_build_974389 -warnings=none -build-cache /tmp/arduino_cache_679110 -prefs=build.warn_data_percentage=75 -verbose /home/lfmiller/Arduino/raspirobottestlibrary/raspirobottestlibrary.ino
/home/lfmiller/.arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/lfmiller/.arduino-1.8.5/hardware -hardware /home/lfmiller/.arduino15/packages -hardware /home/lfmiller/Arduino/hardware -tools /home/lfmiller/.arduino-1.8.5/tools-builder -tools /home/lfmiller/.arduino-1.8.5/hardware/tools/avr -tools /home/lfmiller/.arduino15/packages -built-in-libraries /home/lfmiller/.arduino-1.8.5/libraries -libraries /home/lfmiller/Arduino/libraries -fqbn=RaspberryPi:piduino:bplus:SerialDestination=disabled,GDBDebug=disabled -ide-version=10805 -build-path /tmp/arduino_build_974389 -warnings=none -build-cache /tmp/arduino_cache_679110 -prefs=build.warn_data_percentage=75 -verbose /home/lfmiller/Arduino/raspirobottestlibrary/raspirobottestlibrary.ino
Using board 'bplus' from platform in folder: /home/lfmiller/Arduino/hardware/RaspberryPi/piduino
Using core 'piduino' from platform in folder: /home/lfmiller/Arduino/hardware/RaspberryPi/piduino
Warning: Board mighty-1284p:mighty-1284p:mighty_int8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_INT8
Warning: Board mighty-1284p:mighty-1284p:mighty_int8_opt doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_INT8_OPT
Warning: Board mighty-1284p:mighty-1284p:bobuino doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_BOBUINO
Warning: Board mighty-1284p:mighty-1284p:mighty_opt8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_OPT8
Warning: Board mighty-1284p:mighty-1284p:mighty doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY
Warning: Board mighty-1284p:mighty-1284p:mighty_opt doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY_OPT
Warning: Board mighty-1284p:mighty-1284p:mighty8 doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_MIGHTY8
Warning: Board mighty-1284p:mighty-1284p:avr_developers doesn't define a 'build.board' preference. Auto-set to: MIGHTY-1284P_AVR_DEVELOPERS
Detecting libraries used...
"/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10805 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/cores/piduino" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/variants/bplus" "/tmp/arduino_build_974389/sketch/raspirobottestlibrary.ino.cpp" -o "/dev/null"
"/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10805 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/cores/piduino" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/variants/bplus" "-I/home/lfmiller/Arduino/libraries/Robot" "/tmp/arduino_build_974389/sketch/raspirobottestlibrary.ino.cpp" -o "/dev/null"
"/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10805 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/cores/piduino" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/variants/bplus" "-I/home/lfmiller/Arduino/libraries/Robot" "/tmp/arduino_build_974389/sketch/raspirobottestlibrary.ino.cpp" -o "/tmp/arduino_build_974389/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/tools/compat/include" -D_GNU_SOURCE -O3  -c -MD -pipe -w -x c++ -E -CC -w -DARDUINO=10805 -DARDUINO_RASPI -DARDUINO_ARCH_PIDUINO  -DRASPI "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/cores/piduino" "-I/home/lfmiller/Arduino/hardware/RaspberryPi/piduino/variants/bplu [... cut ...]
arduino.cc/builder.(*Builder).Run(0xc4202bbd28, 0xc420034c00, 0xc420045d88, 0x562240)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 +0xb9c
arduino.cc/builder.RunBuilder(0xc420034c00, 0xc420045d88, 0x64d248)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 +0x99c
/home/lfmiller/.arduino-1.8.5/arduino-builder returned 2
Error compiling for board RaspberryPI B+/2.
```