me-no-dev/RasPiArduino #98: Windows Arduino IDE compilation error: arm-linux-gnueabihf-ar: `u' modifier ignored since `D' is the default (see `U')

starter2019, 2019-03-08:
When compiling a simple code before generating a binary, I got an error on Arduino IDE: 

C:/SysGCC/Raspberry/bin/arm-linux-gnueabihf-ar: `u' modifier ignored since `D' is the default (see `U') 

[image]
Could you please help me ?﻿

starter2019, 2019-03-09:
Any answer here please ?

Barabba11, 2019-03-09:
looking the time they released the last version.. I don't think they are developing it more and fixing issues. You better think to move to system like Node Red, for example. After loosing 2-3 weeks here and try to make all my project work I moved out

powhound4, 2019-04-02:
I too am getting the same error, when I go to compile a simple arduino sketch after I have installed the gnu tool chain and piduino following the tutorial. The error is:

C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-ar: u' modifier ignored since D' is the default (see `U')

This prints about 10 time during compiling.

Any help on how to fix this would be much appreciated!