me-no-dev/RasPiArduino #106: Network libraries and network examples do not seem to be suitable...

rin67630, 2020-07-06:
I have installed successfully and compiled the first sketches without flaws.

However the built-in examples UDPSendReceiveString and UDPNTPClient refuse to work.
The used Ethernet libraries seem to be the original ones from Arduino, intended to drive an Arduino shield.

But we here have a Linux routine running inside Raspbian and the internet connection should come from the OS, not from an Arduino shield.
Could it be possible to provide working examples for this?

bmitov, 2020-08-10:
I am in process of adding support for Raspberry Pi in Visuino, and as part of this, I am writing Arduino compatible Client, Server and UDP sockets.
If there is interest, once I am done, I am offering them for inclusion in this project :-)

rin67630, 2021-07-17:
Hi sorry for the late response.
Yes, I am interested since I am making a new attempt to compile for the Raspberry Pi.
I suppose that there is no need to run a WifI library since the connection is already there. 
Can it be that we don't need UDPSendReceiveString and UDPNTPClient neither since the time is also there and UDP is already completely supported?