me-no-dev/RasPiArduino #109: Compile errors with ArduinoJson

plop510, 2020-11-08:
I recently discover this repository and tested it with diferent arduino or ESP programs and I´m really impressed and thankfull for this excelent job that open paths to many posibilities.

The thing I´m testing now is I2C master (RPI) / slave (Arduino) using JSON over the I2C interface by means of ArduinoJson library. Any example (as master) that compiles for Arduino or ESP32, doesn´t compile for RPI: just on the first includes: 
#include <ArduinoJson.hpp>, #include <ArduinoJson.h>

I get a bunch of errors like that:
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/FlashStringIterator.hpp:14:12: error: 'ArduinoJson6170_91::FlashStringIterator::FlashStringIterator(const char*)' cannot be overloaded with 'ArduinoJson6170_91::FlashStringIterator::FlashStringIterator(const __FlashStringHelper*)'
   explicit FlashStringIterator(const char* ptr) : _ptr(ptr) {}
            ^~~~~~~~~~~~~~~~~~~
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/FlashStringIterator.hpp:11:12: note: previous declaration 'ArduinoJson6170_91::FlashStringIterator::FlashStringIterator(const __FlashStringHelper*)'
   explicit FlashStringIterator(const __FlashStringHelper* ptr)
            ^~~~~~~~~~~~~~~~~~~
....
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/FlashStringAdapter.hpp:56:27: error: ambiguating new declaration of 'ArduinoJson6170_91::FlashStringAdapter ArduinoJson6170_91::adaptString(const __FlashStringHelper*)'
 inline FlashStringAdapter adaptString(const __FlashStringHelper* str) {
                           ^~~~~~~~~~~
.....
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6170_91::enable_if<ArduinoJson6170_91::is_pointer<T>::value, T>::type ArduinoJson6170_91::pgm_read(const void*) [with T = const char*; typename ArduinoJson6170_91::enable_if<ArduinoJson6170_91::is_pointer<T>::value, T>::type = const char*]':
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/DeserializationError.hpp:103:12:   required from here
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:42: error: 'pgm_read_ptr' was not declared in this scope
   return reinterpret_cast<T>(pgm_read_ptr(p));
                              ~~~~~~~~~~~~^~~
C:\Users\plop5\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:42: note: suggested alternative: 'pgm_read_word'
   return reinterpret_cast<T>(pgm_read_ptr(p));
                              ~~~~~~~~~~~~^~~
                              pgm_read_word
exit status 1
Error compilando para la tarjeta RaspberryPI B+/2.

Etc. Please, can you help me to go ahead? maybe some compile option setting...

Thanks in advance,

pfink, 2021-01-16:
This could be fixed by setting these options:

```
-D ARDUINOJSON_ENABLE_ARDUINO_STRING
-D ARDUINOJSON_ENABLE_PROGMEM=0
```

plop510, 2021-01-20:
Thanks pfink, I´ll test it.

plop510, 2021-01-20:
Hello pfink,

I put a file: platform.local.txt on ..\arduino-1.8.9\hardware\RaspberryPi\piduino\platform.local.txt
with just this line:
  "compiler.cpp.extra_flags=-DARDUINOJSON_ENABLE_ARDUINO_STRING -DARDUINOJSON_ENABLE_PROGMEM=0"

Now, when compiling, I don´t get the previous errors but the result is different: I get 20 error lines with this content:
  "C:/SysGCC/Raspberry/bin/arm-linux-gnueabihf-ar: `u' modifier ignored since `D' is the default (see `U')"

and no code is generated.

Please help.

pfink, 2021-01-20:
Sorry don't know how to solve it I'm building with platformio.

plop510, 2021-01-21:
Hello pfink,

Please, forget my last message. Your solution worked. At least RPI code is generated OK.
I will test It.

Thanks a lot.

Best regards,